4ms/metamodule #359: API Change: Add `float CoreProcessor::get_param(unsigned id)`

danngreen, 2024-09-04:
Having this as part of the CoreProc interface would solve a few problems, including #324 and #358

`SmartCoreProcessor` and `VCVModuleWrapper` already keep the param values in an array of floats, so implementing this would be trivial for those. Non-smart CoreProcessor modules would need to store the param values, which would be easy (in some cases they're already stored).

rjsmith, 2024-09-22:
Apologies for posting directly on this issue (is this OK, or better to start related thread in the MM forum?) 

If you are going to change the API, you could extend this to return a struct of related parameter values:

- the raw float parameter value
- the display value (Rack modules can report separate display values, for more meaningful display on UIs)
- the parameter name (one day, this would allow MIDI output transmission of rich parameter feedback to capable controllers)

Exposing the display value might make it possible for the MM UI to show the reported display value instead of a percentage ?  (e.g. "-5.5 db" instead of "34 %)

If not, then maybe at least change the float - returning method name to "get_param_value". (a parameter is more than just its internal float value).

danngreen, 2024-09-23:
This is a great place for discussion about issues from a code/technical perspective, so your post is welcome!

> If you are going to change the API, you could extend this to return a struct of related parameter values:
>
> the raw float parameter value
> the display value (Rack modules can report separate display [...]

Yes, displaying the "display value" would be useful, but `CoreProcessor` is not the place for that since its solely for the audio (DSP) engine. It's the GUI, not `CoreProcessor`, that wants to know about things like parameter names and display values. Here's how I'm implementing that in v2.0:
https://github.com/4ms/metamodule-core-interface/blob/82eb11f774bef89ecb834f61b99adfd9d8344e56/CoreModules/elements/base_element.hh#L68-L78

Similar to how VCV Rack stores the units and scaling info in the `ParamQuantity` (as opposed to the `Param` itself), MetaModule will store this in the `Element`  variant types (as opposed to the data structures of the module itself).

The parameter name is already available through this interface. All non-null elements inherit from `BaseElement` which has a field for `short_name` (and one for `long_name` which we aren't taking advantage of yet):
https://github.com/4ms/metamodule-core-interface/blob/82eb11f774bef89ecb834f61b99adfd9d8344e56/CoreModules/elements/base_element.hh#L36-L54

The GUI has access to the list of elements, which it can look up from the module registry (confusingly called "ModuleFactory" for historical reasons), and retrieve whatever fields it needs. You can see how the GUI is already using the parameter name to display the list of elements if you go to the Module View page. Each of those items in the roller were queried from the list of elements. The XY position, the element type, the image needed to draw the element, etc. are all in that list of elements and that's how the GUI knows how to draw and display a module.

> If not, then maybe at least change the float - returning method name to "get_param_value". (a parameter is more than just its internal float value).

Here is the CoreProcessor wrapper that VCV-rack ported modules use:
https://github.com/4ms/metamodule-rack-interface/blob/v2.0-dev/include/metamodule/VCV_module_wrapper.hh

I picked `get_param()` because it mirrors `set_param()`. As far as CoreProcessor is concerned, a param is just a float (the GUI cares about names, units, etc., but the audio engine just runs the DSP code).

danngreen, 2024-10-30:
This API change is done in v2.0-dev. 
There are some 4ms modules left to migrate, but all modules based on `VCV_module_wrapper` (e.g. `rack::Module`) and `SmartCoreProcessor` work out of the box.

closed 2024-10-30 by danngreen