4ms/metamodule #364: patch causes screen to freeze on load

offthesky, 2024-09-05:
when i load the attached patch, the metamodule screen freezes and i need to hard reboot the whole skiff.  this happens consistently with this patch on both of my metamodules. i would expect the metamodule to just throw the usual "audio stop... cpu >99%" message, but not lock the module up.

[BOGfdbkLPG2x.zip](https://github.com/user-attachments/files/16882453/BOGfdbkLPG2x.zip)

also i'm curious: is there a way to soft reboot the mm when the screen freezes like this? that way if one goes down, don't have to repower the whole skiff.

danngreen, 2024-09-05:
Hmm, OK I can try this patch out with a debugger attached and see. I do see 47 modules in the patch, but it should just do what you say (display an error and not play).

There's no soft reboot, well, there's a reset button if you can access the back of the module but that's not likely.
I think we could have it reboot if it hits a fault like that

offthesky, 2024-09-05:
having the module reboot on hang would be really useful, as we all push the mm limit in learning what it can cannot do. thank you!! also very great info about the reset button on the back. is that a full factory reset though?

danngreen, 2024-09-05:
The reset button on the back is just power off and back on. Not "Reset all settings" or anything like that

danngreen, 2024-09-05:
OK I fixed this, it'll be in v1.1.1.
The problem was the number of modules. It now returns an error if you try to load more than 32 modules in a patch

And yeah, I raised the limit to 64 just to see, and there's no way that patch of 47 modules will play!

closed 2024-09-05 by danngreen