4ms/metamodule #365: patch gives "too much cpu" error on load, but then always loads just fine right after

offthesky, 2024-09-05:
with system set to 48/512, load the patch. it should throw the "audio stopped due >99%" error. load the patch a second time immediately after, it should load/work just fine clocking in around 75-80%cpu, ive been able to repro this on both of my metamodules.  

also i noticed this patch sometimes sends a really loud noise burst through the reverb on load, so loud that the reverb takes over a minute to cool off/the sound to disapate. and other times it doesn't cause this burst on load.  the burst never happens in desktop vcv.  would love to know what might cause this problem, if it's a bug, or somehow just my poor patching ability (but again, it only happens maybe 50% of the time so i feel there could be bug at play here)

[EnterTheVoids.zip](https://github.com/user-attachments/files/16882528/EnterTheVoids.zip)

offthesky, 2024-09-05:
i think i may have pinpointed the load spikes to patches that use a feedback array with bog cvd delays. i have another patch that is basically just cvds and vcv mix modules - multiple loopers. it clocks in max 60%, yet exhibits the same patch load spike behavior. and it also seems overloads when i turn a meta knob that is mapped to the delay time on all the cvds.  when i have time i'll do more tests to try and better verify this issue with another patch example posted here.

danngreen, 2024-09-05:
Ok, yeah I'm looking at the CVD code and it does some dynamic allocation in the "hot" loop the first time it gets called. It sets up the delay engine, which spikes the CPU. I think some other Bogaudio modules do this too?

So this probably should be addressed in the Bogaudio code...

But -- I did just make a workaround on the MetaModule which just runs each module once when you load the patch. We'll need to test this a bit more, it could have unintended consequences. But it does work for this patch (starts right away, no huge noise).
Here's that version: https://github.com/4ms/metamodule/releases/download/firmware-v1.1.1-fix1/metamodule-firmware-v1.1.1-fix1-firmware-assets.zip

Nice patch, too BTW!

offthesky, 2024-09-06:
thanks! really appreciate all your help with this one(and all the other issues)!!! finger's crossed i can perform with these patches ok at an upcoming event early oct.  so i've tried out this new 1.1.1 fix 1 and im still get those loud pops at the intro at times, and also random spikes/audio shutoff now while playing enterthevoids one, it seems to happen when i turn different knobs (for example the feedback input, which is just the vcv mix module)

attached is another tester that also runs a lot of cvd's and vcv mixes, with that air windows verb. this clocks around a moderate 67%, yet i still get random spikes/audio shutoff while turning knobs related to the vcv mix module.

[DualLooperFX+Vrb.zip](https://github.com/user-attachments/files/16901461/DualLooperFX%2BVrb.zip) - in at just under 32 modules ;]

danngreen, 2024-09-06:
Ok, wow I suppose I should raise the module limit to 64. I didn't think any interesting patch with 32 modules would run. But I guess I was wrong!

Regarding this patch and clicks, I think this is due to the modules themselves because I get far fewer issues with block size at max. There's nothing the MetaModule can do about a module's code taking too long to do its processing. On a host computer this often evens out or is just a click, but the MetaModule has hard real-time requirements, which it needs to in order to guarantee a particular latency. We'd need to modify the plugin's code to have it do intensive tasks up front (like allocating delay lines), instead of during audio processing.

Re: long delays.. have you tried the DLD? It has two channels with up to 3 minutes each.

offthesky, 2024-09-06:
you won't find me complaining about a higher module limit in mm, although if we eventually see something like stackable inputs, (or maybe that prototype module within mm), will probably never come close to the limit then

also thank you for that tip- i didn't realize that about the DLD, very nice! time to dig into some manuals (for all the 4ms ones really) and reconfig this looper patch

re patch/clicks/improving plugins- down to test out whenever wherever needed there!

danngreen, 2024-09-06:
Cool, I'm going to close this here since there's nothing to be done in the firmware about this. If we can narrow down which module or set of modules cause issues, then we can open up an issue in the [metamodule-plugin-examples](https://github.com/4ms/metamodule-plugin-examples/issues) repo (or report in the forum, that's fine too). Just trying to stay organized!

closed 2024-09-06 by danngreen