4ms/metamodule #375: Add bypass option for modules

danngreen, 2024-09-10:
Add a button/toggle on the ModuleView page.

Display bypassed module differently on the PatchView (dimmed?)

At first, just mute outputs and ignore inputs. Then, we will extend module info format to include a table of bypass routes. Once that happens we use the table.

hemmer, 2024-09-10:
ICYMI Rack API encourages developers to set this in their modules, e.g. https://github.com/hemmer/Befaco/blob/v2/src/SpringReverb.cpp#L87

danngreen, 2024-09-12:
I started implementing bypass on the 4ms VCV modules by just using `configBypass`. The bypass routes will be in the 4ms CoreModules info structs, so they'll be available to firmware.
To cross-reference, here's the issue for that:
https://github.com/4ms/4ms-vcv/issues/14

danngreen, 2024-09-15:
Ok, just some notes... If someone saves a patch in VCV which has some bypassed modules, then those will need to be bypassed when the patch is loaded. So "bypassed" needs to be added to the patch format, per-module

I have been trying to think of a way around changing the CoreProcessor API since changes here would bump up the Plugin SDK API version. But I haven't figured anything out. At this point the most simple idea I've come up with is to add a new member variable `bool bypassed` to CoreProcessor, which isn't a big change but we'll want to roll that out at the same time as other API changes.

danngreen, 2025-03-07:
`CoreProcessor::bypass` is present in v2.0 api. 
The way forward is for modules to implement something like:
```c++
void update() {
    if (bypass) {
      out1 = in1;
      out2 = in2;
      return;
    }
    //... rest of update() code
}
```

danngreen, 2025-04-22:
Just updating on this, VCV-ported modules will automatically have `processBypass` called if they are bypassed. However there is currently no UI to bypass a module, and bypass state is not stored in the patch file, so the feature is not useful yet. But the backend API for it is part of v2.0, so adding this feature will not require plugins to be re-compiled