4ms/metamodule #491: Support Rotary Switches (snapping Knobs)

danngreen, 2025-04-09:
Examples: Befaco EvenVCO octave knob, Valley InterZone LFO WAVE knob.

These are currently drawn properly with the knob snapping to the correct positions.
However, we could add support for two more things when manually adjusting (using the `Adjust...` pop-up)
- Labels for each snap position. Currently, the integer value is displayed. It would be nice to see text labels like we do for flip switches and slide switches.
- Manually adjusting such a knob requires you to spin the rotary until it snaps to the next position. It would be nice if it would jump automatically to the next position with each rotary click (again, like switches do).

To make this work, we could add a new Element type: RotarySwitch. I'm not sure if it's best if it derives from Knob or from Switch, but it would combine behaviors of both: Drawn like a knob (single SVG that's rotated), but with labelled positions like a switch.

Ericxgao, 2025-04-26:
I'd like to give this a try. 

I have a working prototype but I'm curious on why a new element type was needed - the pot element has enough info as is to derive the snapping behavior, as you can see if `integral=true` and then figure out how many discrete steps the knob has from its `min` and `max`

Ericxgao, 2025-04-26:
https://github.com/4ms/metamodule/pull/499